tactcomplabs/rev #361: RevTracer should be OFF by default

jleidel, 2025-01-13:
**Describe the bug**
RevTracer should be OFF by default.  On the current devel branch its enabled by default which causes issues in debug if the Spike simulator is not installed.  Also, it appears there are bugs parsing the `trc*` SST parameters from RevCPU.cc.  The underlying tracing logic prohibits the code to reach a point where debugging info is available

```
-DREV_TRACER=OFF
```

**To Reproduce**
cmake -DCMAKE_BUILD_TYPE=Debug -DREV_ENABLE_TESTING=ON -DRVCC=/scratch/riscv/rv64-13.2.0/bin/riscv64-unknown-elf-gcc -DRVCXX=/scratch/riscv/rv64-13.2.0/bin/riscv64-unknown-elf-g++ ../

**Expected behavior**
A clear and concise description of what you expected to happen.

**Trace**
If applicable, post a trace of the execution.  

**Additional context**
Add any other context about the problem here.

leekillough, 2025-01-13:
> On the current devel branch its enabled by default which causes issues in debug if the Spike simulator is not installed.

What is wrong with the tracing when Spike is not used?  Is the `.SetMnemonic()` not setting the mnemonics for instructions correctly, causing a wrong disassembly? I can fix `.SetMnemonic()` calls in the instruction tables if they are not being printed correctly for Forza or other.

jleidel, 2025-01-13:
output traces are broken

jleidel, 2025-01-22:
I can try to reproduce tomorrow on gizmo w/ a Forza test

jleidel, 2025-01-22:
I can't seem to reproduce on gizmo... closing

closed 2025-01-22 by jleidel